hanson-young/nniefacelib #1: Hi3559AV100R001C02SPC030编译可执行文件失败，是否需要特殊配置？<make -f Makefile.Debug>

jingruhou, 2020-03-11:
`aarch64-himix100-linux-gcc -Wall -g -I/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../common -I/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../../../mpp/out/linux/multi-core/include -I/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../audio/adp -I/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../../../osal/include -I/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../../../drv/extdrv/tlv320aic31 -I/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../../../drv/extdrv/motionsensor/motionsensor_mng -I/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../../../drv/extdrv/motionsensor/motionsensor_chip/pub -Dhi3559av100 -DHI_XXXX -lpthread -lm -ldl -DISP_V2 -lstdc++ -DSENSOR0_TYPE=SONY_IMX477_MIPI_8M_30FPS_12BIT -DSENSOR1_TYPE=SONY_IMX477_MIPI_8M_30FPS_12BIT -DSENSOR2_TYPE=SONY_IMX477_MIPI_8M_30FPS_12BIT -DSENSOR3_TYPE=SONY_IMX477_MIPI_8M_30FPS_12BIT -DSENSOR4_TYPE=SONY_IMX277_SLVS_8M_30FPS_12BIT -DSENSOR5_TYPE=SONY_IMX277_SLVS_8M_30FPS_12BIT -DSENSOR6_TYPE=SONY_IMX277_SLVS_8M_30FPS_12BIT -DSENSOR7_TYPE=SONY_IMX277_SLVS_8M_30FPS_12BIT -DHI_ACODEC_TYPE_INNER -DHI_ACODEC_TYPE_HDMI -Wall -fno-aggressive-loop-optimizations -ldl -ffunction-sections -fdata-sections -O2 -fstack-protector-strong -fPIC -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE   -Wall  -DHI_RELEASE -Wno-error=implicit-function-declaration -DVER_X=1 -DVER_Y=0 -DVER_Z=0 -DVER_P=0 -DVER_B=10 -DUSER_BIT_64 -DKERNEL_BIT_64 -Wno-date-time -I/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../common -I/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/vision -I/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/sample_nnie_software -O3   -c -o /home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../common/sample_comm_vi.o /home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../common/sample_comm_vi.c
/opt/hisi-linux/x86-arm/aarch64-himix100-linux/host_bin/../target/usr/lib/crt1.o: In function `_start':
**/usr1/code/CODE_CCTD_V1R2C00/code/current/SDK_CPU_UNIFIED/build/script/aarch64-himix100-linux/aarch64_himix100_build_dir/src/glibc-2.24/csu/../sysdeps/aarch64/start.S:84: undefined reference to `main'
collect2: error: ld returned 1 exit status
/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../smp_linux.mak:10: recipe for target 'nnie_face_test' failed
make: *** [nnie_face_test] Error 1**
`

hanson-young, 2020-03-15:
目前我只在HI3516DV300上进行了测试，还不是特别清楚在3559上的运行情况

seanxie28, 2020-04-03:
3516的sample中有些函数在3559中没有，还有makefile也有一些细微差异，按照报错修改可以编译成功。

但是运行时报错 `[Func]:HI_MPI_SVP_NNIE_LoadModel [Line]:435 [Info]:Error(0xa0338003): u32ArchVersion(12) of input model should be 11!`。

看这个意思，请问是否因为3516是ARMv7，而3559是64位的ARMv8，模型不能通用？

hanson-young, 2020-04-03:
@seanxie28 这个意思是NNIE模型的版本不一样，需要重新量化一下
[image]

seanxie28, 2020-04-03:
哦哦，是NNIE的版本，那能否麻烦您也提供一下1.1版本的模型文件呢

hanson-young, 2020-04-03:
caffemodel在目录底下，你可以尝试下自己量化下

seanxie28, 2020-04-03:
好的，我这边尝试之后给您反馈，多谢

closed 2020-05-13 by hanson-young

jstzwjr, 2020-06-20:
> 3516的sample中有些函数在3559中没有，还有makefile也有一些细微差异，按照报错修改可以编译成功。
> 
> 但是运行时报错 `[Func]:HI_MPI_SVP_NNIE_LoadModel [Line]:435 [Info]:Error(0xa0338003): u32ArchVersion(12) of input model should be 11!`。
> 
> 看这个意思，请问是否因为3516是ARMv7，而3559是64位的ARMv8，模型不能通用？

你好，请问需要修改哪些地方？我也遇到了这个错误，undefined reference to main，谢谢

hanson-young, 2020-06-20:
3559和3516的NNIE版本是不同的，因此需要重新生成wk模型