hanson-young/nniefacelib #4: make -f Makefile.Debug出错了

jingruhou, 2020-03-25:
`nniefacelib/vision/sample_svp_nnie_software.c: In function ‘SAMPLE_SVP_NNIE_Cnn_GetTopN’:
nniefacelib/vision/sample_svp_nnie_software.c:2693:9: warning: implicit declaration of function ‘SAMPLE_SVP_NNIE_CONVERT_64BIT_ADDR’ [-Wimplicit-function-declaration]
         pstNnieParam->astSegData[0].astDst[0].u32Stride,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2693:44: error: expected expression before ‘HI_S32’
         pstNnieParam->astSegData[0].astDst[0].u32Stride,
                                            ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2698:44: error: expected expression before ‘HI_S32’

                                            ^~~~
nniefacelib/vision/sample_svp_nnie_software.c:2700:44: error: expected expression before ‘HI_S32’
     SAMPLE_SVP_CHECK_EXPR_RET(HI_SUCCESS != s32Ret,s32Ret,SAMPLE_SVP_ERR_LEVEL_ERROR,
                                            ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c: In function ‘SAMPLE_SVP_NNIE_FasterRcnn_Rpn’:
nniefacelib/vision/sample_svp_nnie_software.c:2779:44: error: expected expression before ‘HI_U32’
         SAMPLE_SVP_NNIE_CONVERT_64BIT_ADDR(HI_S32,pstSoftwareParam->stRpnBbox.u64VirAddr),
                                            ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2780:44: error: expected expression before ‘HI_S32’
         &pstSoftwareParam->stRpnBbox.unShape.stWhc.u32Height);
                                            ^~~~~~
In file included from /home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../../../mpp/out/linux/multi-core/include/hi_comm_svp.h:16:0,
                 from nniefacelib/vision/sample_svp_nnie_software.h:8,
                 from nniefacelib/vision/sample_svp_nnie_software.c:1:
/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../../../mpp/out/linux/multi-core/include/hi_type.h:79:33: error: expected expression before ‘void’
 #define HI_VOID                 void
                                 ^
nniefacelib/vision/sample_svp_nnie_software.c:2783:44: note: in expansion of macro ‘HI_VOID’

                                            ^
nniefacelib/vision/sample_svp_nnie_software.c: In function ‘SAMPLE_SVP_NNIE_Pvanet_Rpn’:
nniefacelib/vision/sample_svp_nnie_software.c:2821:5: warning: implicit declaration of function ‘SAMPLE_SVP_NIE_PERF_STAT_DEF_VAR’ [-Wimplicit-function-declaration]

     ^
nniefacelib/vision/sample_svp_nnie_software.c:2824:5: warning: implicit declaration of function ‘SAMPLE_SVP_NNIE_PERF_STAT_BEGIN’ [-Wimplicit-function-declaration]
     s32Ret = SVP_NNIE_Rpn(pstSoftwareParam->aps32Conv,pstSoftwareParam->u32NumRatioAnchors,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2833:78: error: expected expression before ‘HI_U32’
         SAMPLE_SVP_NNIE_CONVERT_64BIT_ADDR(HI_S32,pstSoftwareParam->stRpnBbox.u64VirAddr),
                                                                              ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2834:44: error: expected expression before ‘HI_S32’
         &pstSoftwareParam->stRpnBbox.unShape.stWhc.u32Height);
                                            ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2837:5: warning: implicit declaration of function ‘SAMPLE_SVP_NNIE_PERF_STAT_EN ’ [-Wimplicit-function-declaration]
     SAMPLE_SVP_NNIE_PERF_STAT_RPN_OP_TIME();
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../../../mpp/out/linux/multi-core/include/hi_comm_svp.h:16:0,
                 from nniefacelib/vision/sample_svp_nnie_software.h:8,
                 from nniefacelib/vision/sample_svp_nnie_software.c:1:
/home1/houjr/hjr/Hi3559A_SDK/Hi3559AV100R001C02SPC030/01.software/board/Hi3559AV100_SDK_V2.0.3.0/mpp/sample/svp/multi-core/nnie/../../../../../mpp/out/linux/multi-core/include/hi_type.h:79:33: error: expected expression before ‘void’
 #define HI_VOID                 void
                                 ^
nniefacelib/vision/sample_svp_nnie_software.c:2842:44: note: in expansion of macro ‘HI_VOID’

                                            ^
nniefacelib/vision/sample_svp_nnie_software.c: In function ‘SAMPLE_SVP_NNIE_FasterRcnn_GetResult’:
nniefacelib/vision/sample_svp_nnie_software.c:2944:63: error: expected expression before ‘HI_S32’
     SAMPLE_SVP_CHECK_EXPR_RET(0 == pstSoftwareParam->stRpnBbox.u64VirAddr,HI_INVALID_VALUE,
                                                               ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2956:44: error: expected expression before ‘HI_S32’
         pstNnieParam->astSegData[1].astDst[0].u32Stride,
                                            ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2958:44: error: expected expression before ‘HI_S32’
         pstNnieParam->astSegData[1].astDst[1].u32Stride,
                                            ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2960:44: error: expected expression before ‘HI_S32’
         pstSoftwareParam->stRpnBbox.unShape.stWhc.u32Height,
                                            ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2965:44: error: expected expression before ‘HI_U32’
         SAMPLE_SVP_NNIE_CONVERT_64BIT_ADDR(HI_S32,pstSoftwareParam->stDstScore.u64VirAddr),
                                            ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2966:44: error: expected expression before ‘HI_S32’
         SAMPLE_SVP_NNIE_CONVERT_64BIT_ADDR(HI_S32,pstSoftwareParam->stDstRoi.u64VirAddr),
                                            ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2967:44: error: expected expression before ‘HI_S32’
         SAMPLE_SVP_NNIE_CONVERT_64BIT_ADDR(HI_S32,pstSoftwareParam->stClassRoiNum.u64VirAddr));
                                            ^~~~~~
nniefacelib/vision/sample_svp_nnie_software.c:2968:44: error: expected expression before ‘HI_S32’

`

hanson-young, 2020-03-25:
我手边没有3559的板子，应该是SDK差异上的一些问题，可能你需要将我的代码迁移到你自己的sample里

closed 2020-05-13 by hanson-young

northeastsquare, 2020-08-28:
就是类型转换方式换一下，采用强制转换方式，可以参考3559里面的例子

MINUS-1ONE, 2020-09-01:
是由于`SAMPLE_SVP_NNIE_CONVERT_64BIT_ADDR` 宏没能发挥作用。

解决方案：参考 `Hi3559AV100_SDK_VX.X.X.X/mpp/sample/svp/multi-core/nnie/sample` 目录下的`sample_nnie.c`文件，将 `SAMPLE_SVP_NNIE_CONVERT_64BIT_ADDR(HI_VOID,pstNnieParam->astForwardCtrl[pstProcSegIdx->u32SegIdx].stTskBuf.u64VirAddr)`等使用`SAMPLE_SVP_NNIE_CONVERT_64BIT_ADDR` 宏的地方，均去掉`SAMPLE_SVP_NNIE_CONVERT_64BIT_ADDR`并修改为 `(HI_VOID*)(pstNnieParam->astForwardCtrl[pstProcSegIdx->u32SegIdx].stTskBuf.u64VirAddr)`这样的`(TYPE *)(variable)`的形式。

pangxiaowen, 2020-09-15:
您好，问一下您编译的时候是否遇到过这样的问题呢。undefined reference to `SAMPLE_SVP_NNIE_PERF_STAT_END'。十分感谢