hanson-young/nniefacelib #7: debug

ChenShisen, 2020-03-30:
# Hisilicon Hi35xx sample Makefile
include $(PWD)/../Makefile.param
SMP_SRCS += $(wildcard $(PWD)/../common/*.c)
include $(PWD)/../../$(ARM_ARCH)_$(OSTYPE).mak

刚开始做这个，请问编译时 这三行的文件是什么？（您编译好的可以运行，我也想编译一下，好像环境不一样）

hanson-young, 2020-03-30:
请问是海思的那款型号的芯片，我是在3516上测试的，不同型号的sample会存在一些不同，有必要的话需要自己移植核心源码

ChenShisen, 2020-03-30:
我下载您的源代码，nniefacelib 文件，$(pwd)/../../ 是上层文件，我这边该怎么办，我运行您编译好的代码是可以的（3516cv500），编译环境和运行环境不是同一台机器。编译时缺少文件该怎么办

hanson-young, 2020-03-30:
有没报错信息？

ChenShisen, 2020-03-30:
include $(PWD)/../../$(ARM_ARCH)_$(OSTYPE).mak

这一行报错：nniefacelib/../../_.mak  not found
应该是这两个变量为空：$(ARM_ARCH)  $(OSTYPE) ，请问这个项目要放在sdk的哪个目录下吗？

hanson-young, 2020-03-30:
Hi3516CV500_SDK_V2.0.1.0/smp/a7_linux/mpp/sample/svp/nnie

ChenShisen, 2020-03-30:
好的 我去试一下 多谢指教

ChenShisen, 2020-03-30:
cant create /***/nniefacelib/../../common/sample_comm_vpss.o:Permission denied
请教一下，编译时会重新编译sdk下面的源码吗？没有sdk的修改权限
SMP_SRCS += $(wildcard $(PWD)/../common/*.c)

hanson-young, 2020-03-30:
会编译源码，这些sdk src是去调用so库的，你确定一下你/nniefacelib/../../common路径的权限，有可能是只读权限

ChenShisen, 2020-03-30:
已经编译成功了 ，源代码会报错 ，yolov3那里 ，代码注释掉就编译通过了，data里图片是640*640，用自己的图片也要resize 到640吗，其他尺寸行吗？

hanson-young, 2020-03-30:
yolov3？我记得已经注释掉了。另外就是必须保持量化时候的尺寸

closed 2020-03-31 by hanson-young