hanson-young/nniefacelib #13: MNN移植Android眨眼效果不理想

xuguozhi, 2020-04-02:
Hi，感谢您的分享。我移植到MNN Android设备上，发现关键点抖动不明显了（相比pfld-lite: https://github.com/ailab26/pfld-lite)
但是在闭眼的时候，眼睛部分点没有反应，可能是loss计算时缺少了对attribute的考虑吧。

hanson-young, 2020-04-02:
谢谢反馈，整体的准确率的上升蒙蔽了我的双眼。我会多留意一些细节的，能给一些示例图片吗？，我下一版本打算做一些针对人脸细节loss上的调整

xuguozhi, 2020-04-02:
自拍几张闭眼的照片即可，我也在试试扩增WFLW数据

hanson-young, 2020-04-04:
PFLD原作者发布的安卓程序也有这个问题。查了下原因，数据集中缺乏闭眼的数据

xuguozhi, 2020-04-04:
嗯，看来WFLW对于眼部数据不够多，可能要做数据扩增了

hanson-young, 2020-04-08:
我融合了106个关键点的数据集中闭眼的数据加入训练，解决了这个问题
https://arxiv.org/abs/1905.03469v1
https://facial-landmarks-localization-challenge.github.io/#index

xuguozhi, 2020-04-08:
用300W原始的数据训练68point，对眨眼效果就比WFLW强。

hanson-young, 2020-04-08:
> 用300W原始的数据训练68point，对眨眼效果就比WFLW强。

对的，300W的对闭眼效果好

xuguozhi, 2020-04-09:
I will try later

hanson-young, 2020-04-09:
优化过的模型已经上传，可以体验一下  
https://github.com/hanson-young/nniefacelib/tree/master/PFPLD/models/pretrained
[image]

xuguozhi, 2020-04-09:
300W 的68点是如何扩展成98点的？差别的点如何补充？

hanson-young, 2020-04-09:
300W中选择部分关键点参与loss的计算以及反向传播，具体可以看代码中的WingLoss

xuguozhi, 2020-04-09:
看来老辛苦了

closed 2020-04-13 by hanson-young

HZWHH, 2020-12-02:
> Hi，感谢您的分享。我移植到MNN Android设备上，发现关键点抖动不明显了（相比pfld-lite: https://github.com/ailab26/pfld-lite)
> 但是在闭眼的时候，眼睛部分点没有反应，可能是loss计算时缺少了对attribute的考虑吧。
请问是如何减小抖动的，谢谢！