hanson-young/nniefacelib #24: 关于pretrain model

lychen0824, 2020-05-01:
你好，我看训练代码的pretrain model是model/pretrain/checkpoint_epoch_final.pth，而测试代码也是用的checkpoint_epoch_final.pth，这个checkpoint_epoch_final.pth应该不是在提供的PFPLD-Dataset上训练得到的吧？

closed 2020-05-02 by lychen0824

budaLi, 2020-05-06:
> 你好，我看训练代码的pretrain model是model / pretrain / checkpoint_epoch_final.pth，而测试代码也是用的checkpoint_epoch_final.pth，这个checkpoint_epoch_final.pth应该不是在提供的PFPLD-Dataset上训练了的吧？

您得出的结论是怎么样的呢

hanson-young, 2020-05-06:
PFPLD-Dataset上训练的是checkpoint_robust.pth，海思上的就是这个模型

budaLi, 2020-05-06:
> 对不起，五一躺尸忘了了回复，PFPLD-数据集上训练的是checkpoint_robust.pth，海思上的就是这个模型
感谢回复  顺便提一下 gpu上onnx2caffe时显存爆出的问题我的解决 ：生成prototxt文件后，手动在所有卷积层的convolution_param 增加 engine: CAFFE，再运行之后的生成caffemodel模型的代码，给后面用的人记录下。

lychen0824, 2020-05-06:
> > 你好，我看训练代码的pretrain model是model / pretrain / checkpoint_epoch_final.pth，而测试代码也是用的checkpoint_epoch_final.pth，这个checkpoint_epoch_final.pth应该不是在提供的PFPLD-Dataset上训练了的吧？
> 
> 您得出的结论是怎么样的呢

你好，我后来细看了下， 感觉checkpoint_epoch_final.pth应该是在原始的WFLW上训练的，后来作者又加了闭眼和head pose训练集，所以直接在checkpoint_epoch_final.pth上finetune了

lychen0824, 2020-05-06:
> sorry，五一躺尸忘了回复，PFPLD-Dataset上训练的是checkpoint_robust.pth，海思上的就是这个模型

你好，再请问下，提供的PFPLD-Dataset数据，WFLW部分的head pose标签是用PRNet生成的吗？

lychen0824, 2020-05-06:
> sorry，五一躺尸忘了回复，PFPLD-Dataset上训练的是checkpoint_robust.pth，海思上的就是这个模型

你好，感谢回复，再请问下，提供的PFPLD-Dataset数据，WFLW部分的head pose标签是用PRNet生成的吗？
还有，LAPA数据集是公开集吗，我没有找到这个集^_^

hanson-young, 2020-05-07:
PFPLD-Dataset数据集训练的时候用的是300W-LP数据集里面的pose，没有用PRNet标注的了
LAPA是京东的数据库
https://www.aaai.org/Papers/AAAI/2020GB/AAAI-LiuY.6557.pdf

xuguozhi, 2020-05-09:
robust模型是直接从PFPLD-Dataset 从头训出来的吗？
或者是先在WFLW训练然后再PFPLD-Dataset finetune，我尝试了个小的模型，但效果浮现不了，点抖动的厉害

budaLi, 2020-05-09:
> 健壮模型是直接从PFPLD-数据集从头训出来的吗？
> 或者是先在WFLW训练然后再PFPLD-数据集finetune，我尝试了个小的模型，但效果浮现不了，点一点的厉害

我觉得就是上面老哥说的直接在checkpoint_epoch_final.pth fintune得到的robust，我也试着训练了下，精度还是可以提升的，目前在WFLW数据集上ION 3,132

xuguozhi, 2020-05-09:
> > 健壮模型是直接从PFPLD-数据集从头训出来的吗？
> > 或者是先在WFLW训练然后再PFPLD-数据集finetune，我尝试了个小的模型，但效果浮现不了，点一点的厉害
> 
> 我觉得就是上面老哥说的直接在checkpoint_epoch_final.pth fintune得到的robust，我也试着训练了下，精度还是可以提升的，目前在WFLW数据集上ION 3,132

部署到移动端 这个模型还是比较大的，而且速度慢了些。所以我打算尝试些更轻量的模型结构

lychen0824, 2020-05-09:
> > > 健壮模型是直接从PFPLD-数据集从头训出来的吗？
> > > 或者是先在WFLW训练然后再PFPLD-数据集finetune，我尝试了个小的模型，但效果浮现不了，点一点的厉害
> > 
> > 
> > 我觉得就是上面老哥说的直接在checkpoint_epoch_final.pth fintune得到的robust，我也试着训练了下，精度还是可以提升的，目前在WFLW数据集上ION 3,132
> 
> 部署到移动端 这个模型还是比较大的，而且速度慢了些。所以我打算尝试些更轻量的模型结构

你好，我也在手机上试了下这个NCNN模型，我只要98点，不需要head pose，但是运行速度比预期慢了很多，暂时不知道怎么回事，head pose这个辅助网络可能有影响？

xuguozhi, 2020-05-09:
> 你好，我也在手机上试了下这个NCNN模型，我只要98点，不需要head pose，但是运行速度比预期慢了很多，暂时不知道怎么回事，head pose这个辅助网络可能有影响？

我觉得可以改造下PFLD 0.25X 应该能快不少 https://github.com/polarisZhao/PFLD-pytorch/issues/23