hanson-young/nniefacelib #26: More light-weighted models

xuguozhi, 2020-05-09:
新开一个帖，大家讨论下兼顾性能与效果的更轻量的模型

xuguozhi, 2020-05-09:
可尝试PFLD X0.25版本

budaLi, 2020-05-13:
> 可尝试PFLD X0.25版本

您好 请问您目前的效果如何  是将pfpld更改为0.25试的么

xuguozhi, 2020-05-13:
> > 可尝试PFLD X0.25版本
> 
> 您好 请问您目前的效果如何 是将pfpld更改为0.25试的么

WFLW ION 3.2 左右

budaLi, 2020-05-13:
> > > 可尝试PFLD X0.25版本
> > 
> > 
> > 您好 请问您目前的效果如何 是将pfpld更改为0.25试的么
> 
> WFLW ION 3.2 左右

速度呢 用的pfld0.25还是pfpld0.25

xuguozhi, 2020-05-13:
> > > > 可尝试PFLD X0.25版本
> > > 
> > > 
> > > 您好 请问您目前的效果如何 是将pfpld更改为0.25试的么
> > 
> > 
> > WFLW ION 3.2 左右
> 
> 速度呢 用的pfld0.25还是pfpld0.25

pfpld基础上面改的，速度是robust的4倍

budaLi, 2020-05-13:
> > > > > 可尝试PFLD X0.25版本
> > > > 
> > > > 
> > > > 您好 请问您目前的效果如何 是将pfpld更改为0.25试的么
> > > 
> > > 
> > > WFLW ION 3.2 左右
> > 
> > 
> > 速度呢 用的pfld0.25还是pfpld0.25
> 
> pfpld基础上面改的，速度是robust的4倍

这么快的么  不过更改网络结构后只能从头开始训练了吧  您是直接砍channel做的优化么