hanson-young/nniefacelib #28: HI3559A下量化merge_bn后的RetinaFace模型精度不理想

Willyoung111, 2020-05-13:
我尝试了用目录下的模型进行在HI3559A上进行量化，最后输出的landmake和bbox的精度不太理想。
[image]
实际上板运行后的结果误差更大
[image]
并且最后没有任何输出
[image]
我尝试过使用只删除了crop层未改变顺序的prototxt进行量化没有出现知乎文章中的错误，但是最后的结果是不正确的
想询问一下可能是哪里出了问题？

hanson-young, 2020-05-13:
bbox和landmark本身数据位会比较低，数值比较小，量化精度不高。以为是对于确定的像素做偏移，对最后的结果影响不会太大。你这个应该是在板子上跑的时候哪里有问题，后处理后的结果明显是溢出了

Willyoung111, 2020-05-14:
嗯感谢，有关bbox和landmark的精度问题了解了，但是有关前面说的用merge_bn后的模型去跑的时候没有输出的问题，我去调试查看了最后输出的3个尺度的cls的数据，发现其中后2维的数据都是约为0，没有置信度较高的数据，也就是实际没有检测到人脸。于是我尝试直接用python跑caffe模型，发现输出的数据也差不多，这是为什么呢？

hanson-young, 2020-05-14:
或许可以参考下这个问题，https://github.com/hanson-young/nniefacelib/issues/8

Willyoung111, 2020-05-14:
我之前看过这个，在您的后处理代码中用于判断置信度的是4个通道中的后两个通道，其中前两个通道的数据都在0.99~1之间（除4096后），而后两个通道的数据都在0~0.01之间，理论上来讲的话有效数据应该是后两个通道的数据吧

Willyoung111, 2020-05-14:
问题解决了，是因为我在量化的时候设置了scale的原因，模型本身已经进行过数据的预处理操作，在量化的时候直接设置norm_type为no processing就可以了。

closed 2020-05-14 by Willyoung111