hanson-young/nniefacelib #29: nnie_mapper 转换 mnet.caffemodel 非常慢

genshengye, 2020-05-15:
博主你好，我在使用 nnie_mapper_gpu  转换一个支持YUV420SP的retinafece模型，发现非常慢，6个多小时了还没转换完，请问是什么问题呢？
cfg文件配置：
[prototxt_file] ./../data/det_face/mnet.prototxt
[caffemodel_file] ./../data/det_face/mnet.caffemodel
[batch_num] 256
[net_type] 0
[sparse_rate] 0
[compile_mode] 0
[is_simulation] 0
[log_level] 3
[instruction_name] ./../data/det_face/retinaface_inst
[RGB_order] BGR
[data_scale] 0.0039062
[internal_stride] 16
[image_list] ./../data/det_face/imageList.txt
[image_type] 3
[mean_file] null
[norm_type] 0

过程日志：
begin net parsing....

end net parsing

begin prev optimizing....

end prev optimizing....

begin net quantalizing(GPU)....

**********************************************************
WARNING: file: Inference::computeNonlinearQuantizationDelta  line: 92
data containing only zeros; set max value to 1e-6.
**********************************************************

WARNING: file: Inference::computeNonlinearQuantizationDelta  line: 92
data containing only zeros; set max value to 1e-6.
**********************************************************

WARNING: file: Inference::convBnParamProcess  line: 816
layer[mobilenet0_conv5_group_15] value[0.00000000000000004205] is too small
**********************************************************

WARNING: file: Inference::convBnParamProcess  line: 816
layer[mobilenet0_conv25_group_69] value[0.00000000000002135883] is too small

**********************************************************

WARNING: file: Inference::convBnParamProcess  line: 816
layer[mobilenet0_conv25_group_88] value[0.00000000000016177380] is too small
**********************************************************

WARNING: file: Inference::convBnParamProcess  line: 816
layer[mobilenet0_conv25_group_104] value[0.00000000000021233644] is too small
**********************************************************

WARNING: file: Inference::convBnParamProcess  line: 816
layer[mobilenet0_conv25_group_109] value[0.00000000000001053666] is too small

end quantalizing

begin optimizing....

一直就卡在  begin optimizing.... 了

hanson-young, 2020-05-15:
我转的时候倒没要那么久，今天我把nnie maper工具上传吧！我把配置文件也给放出来，你试试

hanson-young, 2020-05-15:
> 博主你好，我在使用 nnie_mapper_gpu 转换一个支持YUV420SP的retinafece模型，发现非常慢，6个多小时了还没转换完，请问是什么问题呢？
> cfg文件配置：
> [prototxt_file] ./../data/det_face/mnet.prototxt
> [caffemodel_file] ./../data/det_face/mnet.caffemodel [...]

最新跟新的readme.md  
https://github.com/hanson-young/nniefacelib/blob/master/README.md

genshengye, 2020-05-15:
> > 博主你好，我在使用 nnie_mapper_gpu 转换一个支持YUV420SP的retinafece模型，发现非常慢，6个多小时了还没转换完，请问是什么问题呢？
> > cfg文件配置：
> > [prototxt_file] ./../data/det_face/mnet.prototxt
> > [caffemodel_file] ./../data/det_face/mnet.caffemodel
> > [batch_num] 256
> > [net_type] 0
> > [sparse_rate] 0
> > [compile_mode] 0
> > [is_simulation] 0
> > [log_level] 3
> > [instruction_name] ./../data/det_face/retinaface_inst
> > [RGB_order] BGR
> > [data_scale] 0.0039062
> > [internal_stride] 16
> > [image_list] ./../data/det_face/imageList.txt
> > [image_type] 3
> > [mean_file] null
> > [norm_type] 0
> > 过程日志：
> > begin net parsing....
> > end net parsing
> > begin prev optimizing....
> > end prev optimizing....
> > begin net quantalizing(GPU)....
> > WARNING: file: Inference::computeNonlinearQuantizationDelta line: 92
> > data containing only zeros; set max value to 1e-6.
> > WARNING: file: Inference::computeNonlinearQuantizationDelta line: 92
> > data containing only zeros; set max value to 1e-6.
> > WARNING: file: Inference::convBnParamProcess line: 816
> > layer[mobilenet0_conv5_group_15] value[0.00000000000000004205] is too small
> > WARNING: file: Inference::convBnParamProcess line: 816
> > layer[mobilenet0_conv25_group_69] value[0.00000000000002135883] is too small
> > WARNING: file: Inference::convBnParamProcess line: 816
> > layer[mobilenet0_conv25_group_88] value[0.00000000000016177380] is too small
> > WARNING: file: Inference::convBnParamProcess line: 816
> > layer[mobilenet0_conv25_group_104] value[0.00000000000021233644] is too small
> > WARNING: file: Inference::convBnParamProcess line: 816
> > layer[mobilenet0_conv25_group_109] value[0.00000000000001053666] is too small
> > end quantalizing
> > begin optimizing....
> > 一直就卡在 begin optimizing.... 了
> 
> 最新跟新的readme.md,我测试的时候也就10几分钟就好了
> https://github.com/hanson-young/nniefacelib/blob/master/README.md

非常感谢，我再用你这个试试

ibsd, 2020-05-19:
> 最新跟新的readme.md,我测试的时候也就10几分钟就好了
> https://github.com/hanson-young/nniefacelib/blob/master/README.md

您上传的 mnet_inst.cfg 里面 RGB_order 是 RGB，如果是用 BGR 来转换会有影响吗？实测是可以工作的。

hanson-young, 2020-05-19:
只是精度上会受到影响

ibsd, 2020-05-19:
> 只是精度上会受到影响
谢谢回答。
那是不是Mobilefacenet和RetinaFace这2个的caffemodel在转换wk时都应该用RGB？

hanson-young, 2020-05-19:
对呢

closed 2020-05-22 by hanson-young