hanson-young/nniefacelib #33: WFLW数据集问题

hejing-maker, 2020-05-22:
WFLW数据集本身只是一个2D的数据集7500用于训练 2500用于测试，但是大佬在这做了数据增强(旋转，镜像等)，而不是用prnet做3D的数据集？镜像的话会不会关键点会不准呢？求大佬解释，谢谢

hanson-young, 2020-05-22:
刚开始是用prnet标注，并且是数据增强后再标注的，但是prnet对斜着抬头的情况下效果并不是很理想，所以把300wlp的拿过来做mask label的训练

hejing-maker, 2020-05-22:
那你测试的时候只是使用了AFLW（使用2次）和又98点测试集，而不是把每个数据集分为一部分测试一部分训练，有什么原因吗，大佬可以的话给个QQ，方便提问，这样有点麻烦

hanson-young, 2020-05-22:
在代码里有区的

ufoym, 2020-06-03:
很棒的工作！请教：原始的WFLW测试集2500个样例，与PFPLD-Dataset训练集的样例是保证没有交集的么？不知PFPLD-Dataset训练集的构建脚本是否也可以分享一下？非常感谢！

hejing-maker, 2020-06-03:
你如果只是想训练98点的数据集的话，其实作者的方法已经可以用了，而且作者给的数据集已经是98增强之后和增加300w_lp的，，你可以直接训练，我这边是想弄106点的。

ufoym, 2020-06-03:
> 你如果只是想训练98点的数据集的话，其实作者的方法已经可以用了，而且作者给的数据集已经是98增强之后和增加300w_lp的，，你可以直接训练，我这边是想弄106点的。
> […](#)

感谢！

主要是我另自用了原始的WFLW测试集作测试，需确认测试结果是否可靠，所以想看看PFPLD-Dataset训练集的构建过程。

https://github.com/hanson-young/nniefacelib/tree/master/PFPLD/data    里的处理貌似跟PFPLD-Dataset训练集还不太一样，比如#30 提到300W_LP有做翻转增强，但似乎PFPLD/data的代码并没有翻转的部分。

hejing-maker, 2020-06-03:
300wlp中有翻转增强的数据集，但是里面没有翻转增强的真实的landmarks值,只有未翻转的那几个（ibug  hellen等）有landmarks,不知道你能明白不

zener90818, 2020-07-31:
@hejing-maker 您好，请问106点数据集的训练怎么改呢，您有改成功吗

hejing-maker, 2020-07-31:
可以改的，该成功了，而且训练起来了，你要不加我QQ 776248249

xuguozhi, 2020-09-03:
> 776248249

同样求如何将106点数据补充进来