hanson-young/nniefacelib #38: 关于重新转模型结果异常问题

jstzwjr, 2020-06-23:
您好，我用的是3559A板子，重新转wk模型之后传到板子上跑，提取到的人脸向量特征值都很大，都是几千 几万的，请问可能是什么原因？
另外，我看您在知乎的文章里提到特征向量的对比，我简单看了下代码之后，CNN_convert_bin_and_print_featuremap.py和caffe_forward.py好像都是通过caffe模型来计算特征向量的，结果理当是一样的吧？好像并没有对比转换之后的wk模型。
谢谢！
[image]

jstzwjr, 2020-06-23:
我使用ruyi studio对比fc层的输出，cosin 相似度在0.9左右，但是用您工程编译的可执行文件跑，特征向量会很大。
[image]
这个是板子上的输出
blobs fc1:
[0.985107 ,0.515625 ,-0.284668 ,-0.950928 ,2.164795 ,0.600342 ,0.659668 ,-0.004639 ,0.279785 ,0.651367 ,-0.539795 ,-1.184814 ,1.030029 ,0.123779 ,-1.169678 ,1.133057 ,0.676758 ,1.468506 ,0.273682 ,0.524170 ,1.630371 ,-0.473145 ,1.793945 ,1.658203 ,-0.221436 ,-1.042969 ,0.395996 ,-0.793457 ,0.353271 ,0.856934 ,0.284668 ,-1.289551 ,0.738770 ,-0.481689 ,0.139160 ,-0.367188 ,-0.913574 ,-0.365234 ,0.656494 ,1.131836 ,0.615479 ,0.606689 ,0.470215 ,1.714844 ,-0.903076 ,-1.298340 ,1.142578 ,0.470459 ,-0.477783 ,1.327393 ,1.561768 ,-0.728271 ,1.592773 ,-0.092285 ,-1.308838 ,0.680908 ,-0.814453 ,0.657959 ,-3.338135 ,-1.688477 ,0.702393 ,1.382812 ,1.667236 ,-0.435059 ,-1.022217 ,-1.306152 ,-0.387451 ,-1.545654 ,-1.809570 ,-0.052979 ,-1.384033 ,2.481934 ,-1.520996 ,-1.636230 ,-1.981934 ,-1.395752 ,-0.013672 ,1.942383 ,-0.134033 ,-0.328857 ,1.493164 ,-0.546631 ,1.017578 ,-0.102051 ,0.133545 ,0.855469 ,0.980957 ,-1.455322 ,0.434570 ,-1.186279 ,0.141357 ,0.645020 ,-0.841064 ,-0.597168 ,0.798096 ,0.905029 ,-2.059570 ,-0.500244 ,-1.073486 ,1.393799 ,-0.516357 ,-0.637939 ,-0.452881 ,-0.977783 ,0.038574 ,0.767578 ,0.285156 ,0.237061 ,0.897217 ,-1.835449 ,-0.591553 ,0.354736 ,-1.855225 ,-0.457764 ,3.332031 ,0.257080 ,1.033203 ,-0.197510 ,0.465088 ,0.293701 ,-0.172119 ,-1.474121 ,0.218506 ,-0.577393 ,-1.337646 ,0.096191 ,-0.468994 ,0.311523 ,465311.156250 ,-350384.843750 ,503639.562500 ,83767.257812 ,450560.937500 ,-81855.812500 ,26144.251953 ,-130826.398438 ,505841.937500 ,-441232.750000 ,90481.507812 ,65849.132812 ,137810.953125 ,304306.187500 ,517170.875000 ,191004.609375 ,-278894.531250 ,76771.179688 ,-70756.578125 ,-275492.343750 ,310916.343750 ,-421947.406250 ,296094.593750 ,314948.625000 ,16005.333008 ,439791.312500 ,361877.406250 ,452053.656250 ,-91001.664062 ,226726.593750 ,-145769.593750 ,-17331.765625 ,165799.343750 ,482439.593750 ,-34920.597656 ,194519.656250 ,47744.312500 ,356576.562500 ,32912.375000 ,49360.503906 ,-302686.687500 ,-347668.312500 ,319889.406250 ,115162.257812 ,176810.640625 ,205538.593750 ,-363885.625000 ,520920.031250 ,-320754.968750 ,-304306.218750 ,-300164.468750 ,-316612.718750 ,267268.968750 ,263244.375000 ,287868.937500 ,271422.750000 ,333059.656250 ,349509.187500 ,355237.500000 ,337205.187500 ,398854.968750 ,413712.406250 ,419446.531250 ,402998.312500 ,521999.437500 ,481095.218750 ,485233.843750 ,468791.312500 ,4096.940918 ,-507839.593750 ,-85895.125000 ,8240.155273 ,114577.945312 ,86337.250000 ,90481.507812 ,123191.070312 ,135682.953125 ,131652.765625 ,156274.515625 ,139989.906250 ,-279647.906250 ,-263186.406250 ,-291940.593750 ,-275502.781250 ,308396.468750 ,505060.593750 ,296084.375000 ,518142.187500 ,333221.312500 ,390629.187500 ,353693.093750 ,-184130.578125 ,439974.343750 ,374502.593750 ,427669.281250 ,-182561.609375 ,505767.343750 ,481250.062500 ,493463.406250 ,511266.812500 ,45226.500000 ,61677.890625 ,32912.390625 ,49363.503906 ,111018.484375 ,86497.609375 ,172433.375000 ,115153.632812 ,176802.328125 ,193257.703125 ,164498.421875 ,-374861.718750 ,-374443.343750 ,285364.218750 ,279668.531250 ,271412.656250 ,-320756.031250 ,-326372.781250 ,-301796.468750 ,-316644.718750 ,477706.968750 ,264724.218750 ,-36949.457031 ,-382408.718750 ,-452342.031250 ,-436294.781250 ,-214662.468750 ,-449784.843750 ,-518823.031250 ,-522170.843750 ,-497543.468750 ,-132871.718750 ,-57584.058594 ,-41136.128906 ,-36992.437500 ,-53421.566406 ,-123377.007812 ,-130156.257812 ,-102881.445312 ,-111041.695312 ,-189170.015625 ,122496.054688 ,212349.500000 ,-187537.453125 ,-485594.468750 ,-330525.031250 ,296084.406250 ,313636.656250 ,-279632.906250 ,-336916.406250 ,-291940.593750 ,469978.218750 ,-345429.031250 ,514793.093750 ,-357733.593750 ,-340228.656250 ,-411222.031250 ,-393816.593750 ,-431718.593750 ,342325.062500 ,-291610.718750 ,-502330.593750 ,-489323.406250 ,-473815.343750 ,-16465.505859 ,-118800.750000 ,-213091.625000 ,-15568.313477 ,-82258.757812 ,-53442.320312 ,-102753.570312 ,-70081.320312 ,-148050.015625 ,240845.609375 ,-70242.570312 ,-398296.500000 ,267272.031250 ,283716.218750 ,287860.531250 ,271572.281250 ,-320756.656250 ,-304309.843750 ,-291972.468750 ,-316612.343750 ,-386544.593750 ,-327717.406250 ,-365960.093750 ,293434.281250 ,-452342.031250 ,-435798.781250 ,-431750.468750 ,-447259.593750 ,-518135.031250 ,-501687.781250 ,-497543.343750 ,-513994.593750 ,126736.000000 ,-1312.806885 ,-20622.113281 ,-53454.832031 ,-82417.007812 ,-106943.195312 ,-102785.445312 ,-119233.695312 ,-189170.015625 ,19944.675781 ,-172015.328125 ,-185026.828125 ,308391.812500 ,324837.093750 ,296084.406250 ,521428.656250 ,-280695.156250 ,-262228.406250 ,-291941.031250 ,-370884.718750 ,-423253.656250 ,515751.875000 ,-357736.406250 ,-341286.406250 ,-411223.531250 ,-149014.421875 ,-423526.593750 ,-427285.218750 ,-477171.906250 ,-214807.781250 ,-489319.593750 ,-472087.718750 ,-225648.625000 ,187613.062500 ,-28781.923828 ,-147488.312500 ,-82540.882812 ,-65809.382812 ,-94561.570312 ,-131372.125000 ,-148050.640625 ,-78354.421875 ,-160367.390625 ,-143906.359375 ,466695.937500 ,226086.406250 ,485238.906250 ,509936.406250 ,-518135.000000 ,-522167.781250 ,-495782.156250 ,-514439.718750 ,-452118.031250 ,-412278.750000 ,-506118.562500 ,101047.539062 ,-388021.000000 ,-150925.312500 ,-193893.078125 ,-134418.843750 ,-263420.218750 ,-304319.687500 ,-378324.468750 ,-480164.687500 ,-197619.015625 ,515148.625000 ,-234306.453125 ,-250637.484375 ,-189 [... cut ...] 570.312500 ,-279632.531250 ,-262964.375000 ,-293444.625000 ,-267299.625000 ,-30848.140625 ,-12435.418945 ,-17555.576172 ,142548.859375 ,-115282.640625 ,-99893.820312 ,-160489.000000 ,-144066.359375 ,-82261.507812 ,336046.625000 ,-94561.625000 ,-79377.328125 ,-254963.015625 ,-238515.578125 ,-234371.453125 ,-250817.390625 ,201475.984375 ,217923.187500 ,230259.515625 ,205623.171875 ,135682.953125 ,152134.390625 ,156274.515625 ,180786.265625 ,69889.960938 ,86337.195312 ,90481.507812 ,76353.406250 ,]

hanson-young, 2020-06-24:
@jstzwjr CNN_convert_bin_and_print_featuremap.py和caffe_forward.py比较，是为了保证中间结果没问题，量化的中间结果需要和CNN_convert_bin_and_print_featuremap.py生成的进行比较。所以有这个步骤。

量化后的也有比较，低精度大概在98%-100%之间，高精度没啥问题。

jstzwjr, 2020-06-24:
@hanson-young 我量化之后fc1层输出与caffe的输出cosine相似度只有0.9左右，是不是太低了？我量化使用了大概10张图。但是对比id10.bgr和id11.bgr两张图片的结果时，相似度大概是0.98。
另外，我在使用ruyi studio转换您提供的人脸检测模型mnet时，耗时很久，上午10点开始make wk，到下午4点还在optimizing。量化使用了10张250*250的图片，请问是什么原因呢？谢谢！

jstzwjr, 2020-06-24:
> @hanson-young 我量化之后fc1层输出与caffe的输出cosine相似度只有0.9左右，是不是太低了？我量化使用了大概10张图。但是对比id10.bgr和id11.bgr两张图片的结果时，相似度大概是0.98。
> 另外，我在使用ruyi studio转换您提供的人脸检测模型mnet时，耗时很久，上午10点开始make wk，到下午4点还在optimizing。量化使用了10张250*250的图片，请问是什么原因呢？谢谢！

closed 2020-06-24 by jstzwjr

hanson-young, 2020-06-24:
请确保你data层以及一些中间层没问题，mnet那个量化速度确实会比较慢，有GPU建议上GPU

jstzwjr, 2020-06-24:
[image]
我用的应该是gpu吧。
我的配置是这样的，mean是null，scale是1，这样设置对吗？
[image]

jstzwjr, 2020-06-25:
> 请确保你data层以及一些中间层没问题，mnet那个量化速度确实会比较慢，有GPU建议上GPU

@hanson-young 
您好！可不可以帮忙看看我的配置是不是有什么问题？已经快1天1夜了，模型 还没转成功？大概还需要多久啊，一直在optimizing。感谢！

[image]

pangxiaowen, 2020-09-21:
您好，我也有遇到提取到的人脸向量特征值都很大，都是几千几万的，请问您是怎么解决的呢？

jstzwjr, 2020-09-21:
> 您好，我也有遇到提取到的人脸向量特征值都很大，都是几千几万的，请问您是怎么解决的呢？

我的问题是源代码打印的特征向量是512维的，转后的模型输出向量是128维的，越界了

pangxiaowen, 2020-09-21:
感谢提醒，多谢啦