hanson-young/nniefacelib #60: 关于3516CV500上mobilefacenet的推理速度问题

onesnow123q, 2020-11-27:
Hi! 你好，目前的模型转换与推理有几个问题想请教一下！
1. 根据提供的mobilefacenet的caffe版本转化为wk模型之后（使用的是3516CV500的linux版本的nnie_mapper），发现在板子上直接运行速度大概为300ms，这个是正常的吗？
2.选择了Low_bandwidth转换出来的mobilefacenet模型为什么要比原模型还大呢？
3. 根据你的博客转换后，发现最后输出的结果与caffe的相似度只有94%，这可能是什么原因导致的呢？会与nnie_mapper的版本之类的有关系吗？
@hanson-young 谢谢！

hanson-young, 2020-11-27:
1、我没用过3516CV500，如果有NNIE，那么这个速度确实慢了，DV300上1T算力，速度也就10ms，模型转换变大了，我确实遇到过，我听别人说是depthwise的原因，官方用的不是group实现方式的，另外精度也会和这个有关系
2、94%，如果是高精度的话，基本上就是转化的问题了，低精度的话99%算正常

onesnow123q, 2020-11-28:
@hanson-young 非常感谢您回复。

- 关于第一点，我看您的项目中nniefacelib/data/nnie_model/face/mobilefacenet_inst.wk 大小是7M多，这个是根据https://github.com/honghuCode/mobileFacenet-ncnn/tree/feature/mobilefacenet-mxnet2caffe这个项目的模型转换出来的吗？

- 关于第二点，我使用的是低精度的模型，您的意思是结果是94%是正常，还是结果需要在99%以上才算正常呢，因为看你的这篇文章https://zhuanlan.zhihu.com/p/107548509，里面提及无量化的caffe与量化模型在板子上的推理结果相似度有99.1%呢，这个使用的是mobilefacenet-mxnet2caffe这个项目的模型吗？

谢谢！

hanson-young, 2020-11-28:
1、是的
2、不正常

closed 2020-12-17 by hanson-young